fdm-monster/fdm-monster #195: Auto Updateting to newest stable/dev version

Tobikisss, 2022-02-21:
Hey,
update idea: implement automatic updates to the latest stable/ Dev version of fdm-monster to improve the maintain time for the end users.

davidzwa, 2022-02-22:
I think this a good idea! The current method of installing FDM does not support this, but we can work out a better way:

- publish the NPM packages `@fdm-monster/cli`
- Users can initiate the server/client with `fdm-cli init` or alias `fdm init`
- Users can update the server/client with `fdm-cli upgrade` or alias `fdm upgrade`

I do understand that this doesn't include direct auto-update, but this requires a bit of thinking first to get sorted. Do have the idea to replace PM2 with our own background service. Not an easy thing however.

davidzwa, 2022-07-27:
@allcontributors add @tobikiss for ideas maintenance

davidzwa, 2022-07-27:
@allcontributors add @Tobikisss for ideas maintenance

davidzwa, 2022-10-31:
@Tobikisss this feature request has been on my mind for quite a while, and in other projects I've worked it out in a way where it is not a process managed by FDM Monster itself. I must add with great pleasure, because coupling updates to the framework is a bad design in my view.

In general the better solution is called Continuous Deployment, and how I got it to work is using Webhooks and either SSH+scripts or a system that accepts the webhook for you and does the work. If you like we can discuss nice options for this on Discord (similar to Nginx Proxy Manager, things like CapRover or Portainer) 👍🏼 

Closing the issue for now, as this is not on my planning anymore. FDM Monster should not manage itself and any solution should be on a separate repo.

closed 2022-10-31 by davidzwa