fdm-monster/fdm-monster #1670: Printer disappears after placing on dashboard

tideline3d, 2023-04-26:
I have two printers configured right now and when I add one of them to the dashboard, it just disappears.   The UI thinks I've placed it, but it does not show up in its placement. 

Here's a screen recording of it happening.   I've deleted and re-added the Dev1 printer a couple times now and it's still doing this.   The printer is connected and testing fine in the UI (I cut that off in the screen recording).  

Of note, both printers have a 'sort index' of 1 in the printer list screen.  

https://user-images.githubusercontent.com/12903320/234694491-1ab78ed4-3a33-4e3c-a3b6-d6148e4aeaee.mov

ndarrow, 2023-04-26:
I have similar problem but not exactly the same. See attached image. 
I can place any printer in the 16 grid locations seen with printers, but I get the disappearing printer problem in any of the other locations (yellow lines).

[image]

davidzwa, 2023-04-27:
@ndarrow I've found the issue you're experiencing and will solve it immediately. The main problem in this issue is more complex, so will need more figuring out.

davidzwa, 2023-04-27:
@tideline3d as stated in Discord I've found the issue 

Quote from Discord: "It's simply due to removing printers that were placed on a floor. MongoDB does not perform ForeignKey checks so leaves behind orphans (as you correctly stated)"

Will fix asap.

davidzwa, 2023-04-27:
The problems that occur are:
- Any ID can be added as a Printer on a Floor (existing or not)
- Printers can be placed on the same floor location (same coordinates)
- The printer removal action is not cascaded to other models like Floor

davidzwa, 2023-04-27:
Fixed in version `1.3.1-rc1` which is rolling out as we speak.

closed 2023-04-27 by davidzwa

davidzwa, 2023-04-28:
@all-contributors please add @ndarrow for bug

davidzwa, 2023-05-02:
@all-contributors please add @ndarrow for financial

PR #1675 (2023-04-27): Bugfix/1670 printer disappears after placing on dashboard
# Description

Fixes #1670 

The problems that occur are:
- [x] Any ID can be added as a Printer on a Floor (existing or not)
- [x] Printers can be placed on the same floor location (same coordinates)
- [x] The printer removal action is not cascaded to other models like Floor

Feature:
- [x] Fix floor positions after loading floors