fdm-monster/fdm-monster #1686: [AutoDiscovery] Develop MQTT auto-discovery for OctoPrint printers

davidzwa, 2023-04-28:
This will make use of Mosquitto as a broker. I'm currently still designing this feature.

davidzwa, 2023-04-28:
Currently leaning towards installing https://plugins.octoprint.org/plugins/mqtt/ for autodiscovery

davidzwa, 2023-04-28:
Not a good sign https://github.com/OctoPrint/OctoPrint-MQTT/issues/131

davidzwa, 2023-04-28:
Requires HA but drops the need for Mosquitto https://plugins.octoprint.org/plugins/homeassistant/
Requires Mosquitto but is easy to break https://plugins.octoprint.org/plugins/mqttpublish/
Bridges the API to MQTT https://github.com/jneilliii/OctoPrint-MQTTSubscribe

davidzwa, 2023-04-28:
https://cedalo.com/blog/nodejs-mqtt-usage-with-mosquitto-example/
https://iot-blogger.com/mqtt-node-js/

mkevenaar, 2024-04-08:
Help wanted
I have implemented SSDP before, but I experienced issues where it was slow or inconsistent (mostly docker). Another doubt I had with it, when I was working on this for OctoFarm, was that I did not know how to complete the authentication triangle (asymmetric crypto / key sharing).

Why SSDP
It's a protocol that is supported by most networks/routers and which does not require separate brokers or applications (compared to MQTT). Furthermore, OctoPrint itself already has it built in, where as MQTT would require two plugins or a newly written plugin with the downside of having to depend on a plugin (instead of a built-in/pre-packaged feature). This does not mean I do not see the benefits of MQTT and a dedicated broker, it's just that I realize SSDP is a nice method to discover OctoPrints out of the box.

davidzwa, 2024-11-04:
At this moment FDM Monster will not implement auto-discovery using MQTT, SSDP and similar host-installed services. If this changes in the future a new issue should be created as a feature request.

closed 2024-11-04 by davidzwa