fdm-monster/fdm-monster #1916: Investigate: Continuous Printing

davidzwa, 2023-06-12:
Such a nice plugin, would be nice to embrace. Let me know below if this will help you!

mariohoelscher, 2023-06-27:
Hi, 
I use an automation system, which is changing the build plate of my printer automatically. It only uses the gcode to change it. Currently I use the Plugin Continuous Printing for managing my printers, but I would love to use FDM-Monster for my printers, because it is more handy. At the moment I can't use FDM-Monster, because it automatically starts the gcode of the printer and don't let me line up gcodes into the queue of the Continuous Printing Plugin. I would be happy if the plugin would be integrated :)

davidzwa, 2023-06-27:
Nice feedback, I've been studying the plugin and how it works a bit. If I get it correctly it works by moving files into a subfolder. 
This is a bit of work to implement. Furthermore, we have to study the plugin's API to see if it exposes the right endpoints to set up new CP-jobs.

Do you have interest in contributing to make this happen?