fdm-monster/fdm-monster #1948: Bug: importing existing printers can clear out floor position (and possibly enabled flag)

davidzwa, 2023-06-18:
Some properties seem to be reset to default if the printer already exists. Luckily the import still works and small problems can easily be rectified manually afterwards.

The import functionality works for the most part.

mkevenaar, 2024-04-08:
@davidzwa, please investigate if this is still reproducible.

davidzwa, 2024-04-12:
I currently have a reproduction of the two problems that can occur.

## My reproduction steps
I had a setup with MongoDB and 1 floor named **Default Floorrr** (note the triple 'r'). It had 7 printers placed on it.
My export contained a floor named **Default Floor** (floor level 1) with 4 printers on it and a floor named "Familiar Ear" with no printers on it. The export had Floor comparison strategy set to 'floor' meaning it will ignore names, but just compare floor level instead.
I imported twice to see the effects.

After import the four printers were visible on the floor, all other 7 printers were unpositioned ("7 unplaced printer(s)" was shown), but also the Default Floorrr floor was gone itself. This suggests the floor was renamed and made empty.

After the 2nd import attempt, the floor was empty. All 11 printers are marked as unplaced. This I can not explain.

## Bug 1 description
The import should use the floor comparison strategy to update floor properties and insert/overwrite printer positions. If another floor exists with a different floor, it should be left alone.

## Bug 2 description
The import should not wipe the floor 🧹, instead printers should be added to it. 

## Overall experience
The import dialog is put on the printer list, but personally I wanted to see the effects on the grid and I had to navigate back there. The import/export dialog felt in rough shape concerning the (complex) actions it was about to do. It was generally unclear what everything meant during the export and it was unclear what was about to happen during the import.
If the import can have such an impact, the user should be presented with the consequences before executing them. This is an enhancement for the future for a separate issue.

davidzwa, 2024-04-14:
I believe I have found a reproducible test case that highlights the problem.

davidzwa, 2024-04-14:
The problem seems most likely related to SQLite only concerning the floor update wiping a floor, which was unrelated to the YAML import. However, I did also spot quite some issues in YAML import with wrong IDs being forwarded to the service (mostly floor position ids, but also a floor id itself).
I found that the 

One note to take for the future: sqlite floor service does a {...floor, prop1, prop2 } type of update, which means that partials will not pass validation. Maybe its best to go back to the mongodb floor service style: cherry pick one property to update instead. It's just so much less failure prone that way.

davidzwa, 2024-04-14:
I've found another issue: floors merge in sqlite mode, but they are overwritten in mongodb mode. I'm leaning towards overwrite (as long as the final floor state is not empty) as this is more consistent on the domain level. In the future we should offer a merge option, which the yaml service should execute. The database service is not responsible for merging data, nor is the store.

closed 2024-04-14 by davidzwa