fdm-monster/fdm-monster #2002: Printers wont connect until Docker stack restarted

tophattwaffle, 2023-06-27:
Printers when imported or created won't connect until Docker stack is restarted.

1. Import or create a new printer
2. See that the printer won't connect, but "Test Connection" is working as expected.
3. Restart docker stack
4. Printer should now connect

https://github.com/fdm-monster/fdm-monster/assets/6774125/f516cf60-e09a-4e84-bdab-7e6eaa224a42

davidzwa, 2023-06-27:
Thanks for the report! I do have a hunch of what might be going wrong here.
Let me get back to you soon, after I do a repro.

davidzwa, 2023-06-28:
#2001 is now resolved (MongoDB startup hiccup issue) and I suspect that this issue is solved with it. 

The reasoning: without the BOOT_TASKS firing off, nothing will push updates to the frontend. That means no updates, but data is always stored in the database. This will now not happen anymore.

closed 2023-06-28 by davidzwa