fdm-monster/fdm-monster #2015: Unraid app setup - needs feedback

davidzwa, 2023-06-28:
There's been a report of someone lacking the resources to set up FDM Monster in Unraid.
If this applies to you, please respond below.
If you feel like contributing on this end, same thing! Help would be greatly appreciated.

cyrixdx4, 2023-09-11:
I use UnRAID as my backend for hosting all my 3d printing files. I used to use OctoFarm for hosting my print farm.  

I am very much in favor for helping in anyway possible for a native docker container image for UnRAID.

davidzwa, 2023-09-11:
So bit of an update about the monolithic build for FDM Monster. A inline feature request on Discord was posted a couple of weeks ago on the topic of monolithic FDMM. And serious work was done to attempt getting there.

One impediment was the database (being MongoDB). I do not like this dependency as I prefer to have a PostgreSQL / SQLite relational setup.

Here's the history:
1) I responded yes to exploring a monolith with SQLite, but only once I could remove MongoDB as I find it to be in the way of a monolithic container as it an unmanaged server dependency. 
2) I tested MikroORM as a fresh start with SQLite, and I found it:
  - nice as it has Unit-of-Work, transactions, simple setup, proper migrations.
  - found it very hard to work with using vanilla JS that FDMM currently uses (sure I could go for `better-sqlite3` or `sqlite3` directly or with `Knex.js`, but that would be a step back in quality)
  - found it MikroORM to be just as nice as TypeORM, even better, if Typescript is assumed.
3) I decided to stop evaluating MikroORM for now.
  - and instead work on the release for 1.5.0 with authentication (user login)
  - prepare to move to a Typescript code base in the future as "reflection" is a big part of the problem causing MikroORM to fail (Step 2).

This means that the monolithic approach either must be:
- with mongodb
- postponed a little

I hope this provides a bit of insight on the monolithic, and hence the state of an Unraid monolithic app.
Of course the split approach (separate mongodb and server) are still good to go.

cyrixdx4, 2023-09-11:
If we need to replace the backend DB that's fine with me.  Can SQLite handle the sheer amount of data being thrown at it and kept around for a longer period of time? I know MongoDB can be overkill...

Monolithic, for me, means that everything is in one place and super easy to support.

davidzwa, 2023-09-11:
This is my strategy (to answer your question):
1) Find a decent ORM
2) Low load case: SQLite mode
3) Medium load case: SQLite or PostgreSQL (default config)
4) High load: PostgreSQL cluster and/or Redis

The load on FDMM's database is not high so far, this is by design and is achievable by working in memory/cache mostly.

davidzwa, 2024-01-16:
With great pleasure I can say that the first `alpha`/`sqlite` containers are rolling off the counter.

The compose file in the docs contains an example how to run it (it's pretty similar to the old way, except that mongodb is not required anymore).

https://docs.fdm-monster.net/docs/installing/docker_compose#step-2-create-a-docker-composeyml-file

closed 2024-03-08 by davidzwa