fdm-monster/fdm-monster #2019: UI freezes after some manipulations in the side bar

WindArrow3d, 2023-06-30:
**Describe the bug**
A clear and concise description of what the bug is.
Sometimes, when using the right-side bar the UI will freeze and stop accepting new mouse clicks. Sometimes it will recover 30 seconds to a minute later. 

**To Reproduce**
Steps to reproduce the behavior:
>It doesn't always work, but after I open the right-side bar, the UI will freeze. 
>It doesn't always work, but after I stop a printer from the right-side bar, the UI will freeze. 
>It doesn't always work, but after I use the right-side bar > Settings > Save, the UI will freeze.
>The UI does sometimes recover on it's own (30 - 60 sec). 

**Expected behavior**
>No freezing

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

>No helpful screen shots of the problem. Side bar is open and mouse curser stops interacting with the page.

**Browser (please complete the following information):**
 - Browser: Chrome Version 114.0.5735.199
 - FDM Monster Client Version: 1.2.4
 - FDM Monster Server Version: 1.4.1
 - FDM Monster Hardware: RPi 4 8GB
 - FDM Monster Hardware OS: MonsterPi 0.2.2

**Additional context**

>There are other bugs that are possibly related that I will be submitting shortly.

WindArrow3d, 2023-06-30:
Additional context

>In my case, I have 2 printers connected and these are where I'm seeing the issues, and 54 printers manually disconnected (no power to printer or pi).

davidzwa, 2023-07-01:
I have a big hunch what is going on here, as I've seen this issue before. When there are API errors the UI seems to freeze as its not properly able to capture and handle the errors/exceptions. This is unacceptable behavior and I will dig deep to fix this asap.

Moving to frontend repo.