fdm-monster/fdm-monster #2020: OctoPrint instances lose connection to FDM Monster

WindArrow3d, 2023-06-30:
**Describe the bug**
A clear and concise description of what the bug is.
Printers will randomly lose connection, and eventually regain connection.

**To Reproduce**
Steps to reproduce the behavior:
>Watch printer tiles and wait. 

**Expected behavior**
>No connection loss. 

**Screenshots**
If applicable, add screenshots to help explain your problem.
>No screenshots of the issue will be helpful. 
[image]

**Browser (please complete the following information):**
Browser: Chrome Version 114.0.5735.199
FDM Monster Client Version: 1.2.4
FDM Monster Server Version: 1.4.1
FDM Monster Hardware: RPi 4 8GB
FDM Monster Hardware OS: MonsterPi 0.2.2

**Additional context**
Add any other context about the problem here.

>There are other bugs that are possibly related that I will be submitting shortly.
>In my case, I have 2 printers connected and these are where I'm seeing the issues, and 54 printers manually disconnected (no power to printer or pi).

davidzwa, 2023-07-01:
I think what you're describing here is the disconnected/powered-down printer tiles. Am I correct?

If so I would propose to change the title to: "printer tiles are toggling continuously, even though nothing seems to be changing and no connection restored".

WindArrow3d, 2023-07-01:
1. Before connection loss: Tile shows "Printing" as normal. 
2. After connection loss: Tile shows "Unset" (see pic).
3. Once connection is restored: Tile shows "Printing" as normal. 

No user inputs from the time the tile shows "Printing", then "Unset", then "Printing" again.  

[image]

davidzwa, 2023-07-03:
I believe you are correct in your conclusions. I have strong belief it's a backend issue related to the feature request in #2028.

Will keep this open as a backend regression! Thank you 🥇

davidzwa, 2023-07-05:
We've worked on this in DM and fixed the issues accordingly. #2051 did contribute albeit only partly.

closed 2023-07-05 by davidzwa