fdm-monster/fdm-monster #2064: Reinstalling removes the systemd service file, needs to reinstall twice (Linux Standalone Service)

doucettom, 2023-07-06:
tdoucet@fdm-monster:~/fdm-monster/installations/fdm-monster-node-linux$ systemctl restart fdmmonster.service 
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
Authentication is required to restart 'fdmmonster.service'.
Authenticating as: tdoucet (tdoucet)
Password: 
==== AUTHENTICATION COMPLETE ===
Warning: The unit file, source configuration file or drop-ins of fdmmonster.service changed on disk. Run 'systemctl daemon-reload' to reload units.
tdoucet@fdm-monster:/etc$ systemctl daemon-reload
==== AUTHENTICATING FOR org.freedesktop.systemd1.reload-daemon ===
Authentication is required to reload the systemd state.
Authenticating as: tdoucet (tdoucet)
Password: 
==== AUTHENTICATION COMPLETE ===
tdoucet@fdm-monster:/etc$ systemctl restart fdmmonster.service 
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
Authentication is required to restart 'fdmmonster.service'.
Authenticating as: tdoucet (tdoucet)
Password: 
==== AUTHENTICATION COMPLETE ===
Failed to restart fdmmonster.service: Unit fdmmonster.service not found.

Reinstalling fdmmonster a second time solves the issue.

Running Ubuntu Server 22.04 and fdmmonster 1.4.4

davidzwa, 2023-07-06:
I've read your bug description. And thanks for checking these steps out! As we discussed in private, the linux scripts are in rough shape. Concerning your bug report, I'm not sure what is causing the service file to disappear, but it is most likely to be related to the uninstall step here on this line (which happens regardless of the bug #2063 you encountered):
https://github.com/fdm-monster/fdm-monster/blob/beb90225e5918ad2ed48a7609bb973b65e0a8102/installations/fdm-monster-node-linux/update-fdm-monster.sh#L55

Due to bug #2063 the service is uninstalled, but no new one is created.

**Actual work to be done**
Currently, the linux service is to be installed and updated with node-linux **within** the cloned repository. This does not make any sense in hindsight, hence I've created issue #1996. This also means the documentation and scripts should be adjusted.
After this work is complete there needs to be a script added to help restart the `node-linux` service.

If you're ok I will repurpose this issue to track the latter steps.

davidzwa, 2024-04-10:
At this point in time Linux installation with custom script has been removed from recommended installation methods.
Until the point it returns, in whatever way or form, this issue will be closed to keep the list of issues focused and clean.

For now this issue has lost relevance and I will close it. Let's reopen it once it the situation asks for it.

closed 2024-04-10 by davidzwa