fdm-monster/fdm-monster #2098: Cannot connect to octoprint instance when using zeroconf/link-local address

rmhenn, 2023-07-12:
**Describe the bug**
Attempting to Octoprint using a .local address results in a "Socket Aborted' error as well as an "API unreachable" error

**To Reproduce**
Steps to reproduce the behavior:

1. Begin Creating Printer
2. Add .local URL to host field, port in port field
3. Click "Test Connection" or "Create"

**Expected behavior**
> Connection is made to Octoprint API successfully

**Screenshots**
[image]

**Browser (please complete the following information):**
 - OS: Windows 10
 - Browser: Chrome
 - Version: 114.0.5735.134

**Additional context**
- Can confirm that connection succeeds when using IPv4 address
- Server 1.4.4 | Client 1.2.8

davidzwa, 2023-07-12:
@all-contributors add @rmhenn for bug
@all-contributors add @rmhenn for ideas

davidzwa, 2023-07-12:
@all-contributors add @rmhenn for bug

wallace11, 2023-07-25:
May I add that using a port should not be mandatory.
For example, I use local reverse proxy, so all my URLs are  `https://pi.local/<PRINTER>`. fdm-monster automatically appends the port to the url which causes it to become incorrect.

davidzwa, 2023-08-13:
Tackling this issue on the frontend as we speak. The backend will follow after.

PR #2181 (2023-08-18): Fix/2098 url validation and normalization
# Description

Should fix #2098 by leaving URLs alone, except for normalizing them w.r.t defaulting to http and slashes, etc.

Needs tests:
- `octopi.local`
- `octopi`
- `https://mysecurepi`
- `localhost`
closed 2023-08-18 by davidzwa