fdm-monster/fdm-monster #2370: Material mangement

Synochy, 2023-09-29:
Hello,
Thanks for the good work !

I would love to have the possibility to do some spool management by assigning it to a printer. 
They will be an initial configuration of each spool (brand, type, quantity of material)
Then for each print the system estimate the quantity of material used and subtract it from the spool assigned. 
Maybe the system could be able to warn you if you are launching a print on a printer with a spool without enough material.

Do not hesitate if you have any question

Thanks by advance,

davidzwa, 2023-09-29:
Hi this is a very good question, asset management is important!

I wanted to bring your attention to https://github.com/Donkie/Spoolman for now.
This is the direction me and @mkevenaar were thinking of bringing FDMM to the next level without taking too much (functional) responsibility on our shoulders.

Is this something that would help you (if integrated into FDMM in a semi-coupled manner)?

Eager to hear your thoughts.

Synochy, 2023-09-30:
Hi,
thank you very much for your answer.
I did not know this tool, I will have a look !
I understand your direction, indeed it could be a good idea to have it integrated or semi-coupled to FDMM)

davidzwa, 2023-10-20:
@Synochy hi there. Would love to hear if that tool Spoolman does the trick for you. That can be enough motivation to start designing an integration for it.

Let me know if you have the chance. Thx!

Simplemente3d, 2023-12-22:
@Synochy there is also a [SpoolManager](https://plugins.octoprint.org/plugins/SpoolManager/) plugin in octoprint, which is quite robust and has the functionality you mention, the drawback being the database lives within the octoprint box.
There are ways to share the database, especially if the octoprint instances reside within the same host.