fdm-monster/fdm-monster #2491: Fail to connect printer on docker-compose localhost

amrsoll, 2023-10-20:
**Describe the bug**

I tried to test FDM by running an Octorpint instance on localhost together with FDM.

I tried adding the OctoPrint instance on localhost

I get ` [API Exception Handler] Error: connect ECONNREFUSED 127.0.0.1:80`

**To Reproduce**
Steps to reproduce the behavior:

1. copy `docker-componse.yaml` here

```yaml
version: '3.4'

services:
  mongodb:
    image: mongo:latest
    container_name: mongodb
    environment:
      - MONGO_INITDB_ROOT_USERNAME=${YOUR_ROOT_NAME}
      - MONGO_INITDB_ROOT_PASSWORD=${YOUR_ROOT_PASSWORD}
    ports:
      - "28017:27017"
    volumes:
      - ./mongodb-data:/data/db
      - ./mongoconfig:/data/configdb
    restart: unless-stopped

  fdm-monster:
    container_name: fdm-monster
    image: davidzwa/fdm-monster:latest
    restart: unless-stopped
    ports:
      - "4000:4000"
    environment:
      - MONGO=mongodb://${YOUR_ROOT_NAME}:${YOUR_ROOT_PASSWORD}@mongodb:27017/fdm-monster?authSource=admin
    volumes:
      - ./fdm-monster/media:/app/media

  octoprint:
    image: docker.io/octoprint/octoprint
    restart: unless-stopped
    ports:
      - 80:80
    volumes:
     - octoprint:/octoprint
```

2.  `YOUR_ROOT_NAME=root YOUR_ROOT_PASSWORD='12345678' docker-compose up`
3.  go to localhost and do the first OctoPrint setup + create an API key
4. Copy API key
5. go to `localhost:4000`
6. Go through 1st time setup
7. create printer using the API key + address `localhost`

**Expected behavior**

> Printer would connect

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Browser (please complete the following information):**
 - OS: Linux (NixOS / docker)
 - Browser Firefox
 - Version 1.5.0 

**Additional context**
Add any other context about the problem here.

amrsoll, 2023-10-20:
might be related to #2181, is it not in the latest stable?

davidzwa, 2023-10-20:
Hi,

Thanks for the bug report. I am also seeing inconsistencies sometimes. Mostly around `octopi.local` mDNS, but it is clear from your case more is going wrong and further investigation/mitigation is needed. 

Just to be sure, what OS are you running on? Are you able to resolve 127.0.0.1, 0.0.0.0, octoprint and/or host.docker.internal as alternative hostnames? 
This might help me track this problem down. 

Also, there's a timeout setting in the settings. This could help, but only partially (I still have to expose the websocket connection setting as well). 

About URL normalization yeah its merged a while ago. Especially for this reason to be honest. It resulted in the current form (on the UI side), which does not split up port and hostname anymore.

I hope you're up to test any fixes or testing elements for this issue that I will implement.

Thanks again!

amrsoll, 2023-10-21:
Hi David,

Thank you for taking the time :D

It could be an issue with my setup, which is why i provided the docker compose to easily recreate fdm + octoprint on the same instance

I had also tried 127.0.0.1 with the same result

Running with docker compose on Nixos.

amrsoll, 2023-10-21:
I tried the following with no luck

- connection timeout 1000 -> 5000 ms, no change (socket aborts instantly)
- http://127.0.0.1 -- http://0.0.0.0 -- http://localhost

I am able to connect to octoprint through the browser to all these addresses. Now I'm wondering if the issue lies in `docker-compose.yaml` in case the containers are not able to communicate between them. It would surprise me given that the mongo server doesn't have any special treatment.

davidzwa, 2023-10-21:
I've tried to reproduce your setup.

```
... previous part was practically the same as yours ... 

  octoprint1234:
    image: docker.io/octoprint/octoprint
    restart: unless-stopped
    ports:
      - 800:80
    volumes:
     - octoprint1234:/octoprint

volumes:
  octoprint1234:
```

Indeed I cannot resolve 0.0.0.0 or 127.0.0.1 (same as localhost) with the test connection method, which tests it from within the FDM Monster container.

I am able to resolve:
- `octoprint1234:80` which is the same as `octoprint1234` - octoprint1234 is a valid hostname inside a docker network
[image]
- `host.docker.internal:800` - with the external docker-compose port, personally I set it to 800
[image]
- `172.20.0.2` - this IP address is what docker auto-assigned to my OctoPrint container. Of course that's not very helpful, but I had to check for the purpose of elimination.
[image]

I will do one quick experiment to check whether the server setup of FDM Monster can make a difference, because the FDM Monster NodeJS setup is bound to 0.0.0.0 inside the container. Maybe that changes things:
[image]

davidzwa, 2023-10-21:
I just checked. Doing a ping and curl to the addresses specified above lead to the following results. 
Note: I had to switch to bash and install iputils and/or curl for this in the FDM Monster container.

Results:
- `127.0.0.1:800` or `127.0.0.1:80` did not work (nor did localhost)
- `0.0.0.0:800` or `0.0.0.0:80` did not work
- `host.docker.internal:800` worked
- `octoprint1234` worked

[image]

Maybe some network configuration needs to be changed, or maybe something needs to be done to ensure `127.0.0.1` is discovered.
Let me know if there is a solution in the options above. In that case this issue can be closed.
If not it would be great to hear what can be done or receive a Pull-Request for changes to the FDM Monster docker container.

amrsoll, 2023-10-23:
Thank you very much @davidzwa I should definitely have been able to find that user error :see_no_evil: at any rate, I can make a test setup and pass the `/dev/ttyX` to the appropriate container to actually have a functional centralised command.

We only have a few printers, so centralising on a semi-decent hardware makes sense for now :ok_hand:

closed 2023-10-23 by amrsoll

davidzwa, 2023-10-23:
Ah thats nice to hear! Best of luck 🖨️

amrsoll, 2023-10-23:
printers go brrr at the makerspace