fdm-monster/fdm-monster #2764: Unraid Docker App Installment

comet424, 2024-01-16:
hi id like to see this App added to the Unraid Server Softwares App Community  to replace the OctoFarm  thats in the community app that isnt supported anymore...  so this can be run on Unraid as a docker

davidzwa, 2024-01-16:
Hi @comet424 that is a good request and you're not the first to ask. 

Could you point me to a method to achieve this? Should there be a Unraid App repository? Would you be interested in contributing to that?

davidzwa, 2024-01-16:
https://github.com/fdm-monster/fdm-monster-unraid/blob/main/README.md

I've created an empty repository for it. Eager to hear what the next steps are.

comet424, 2024-01-16:
@davidzwa
i actually dont know how to make the repositories  i still not 100% good at linux even though i learned it in the 90s yet i went the windows way lol...    all i know is to contact the unraid community  or the discord...

but i can ask in discord or the unraid forums what the steps are to create it.... as i not 100% sure what you do..  i checked your app wasnt there and found out that the octofarm is discontinued.. when i couldnt figure out to run it on unraid lol  with this mono database which was a pain and found its no longer supported... but i can ask  about how you go about adding yours or the steps

davidzwa, 2024-01-16:
No problem, let me know if you need me to join a discord to start the conversation with them directly.

comet424, 2024-01-16:
so i found on github  i also  asked on the forums  but i guess you request a template...
https://github.com/selfhosters/unRAID-CA-templates?tab=readme-ov-file

click on the create a new issue    if that doesnt help  i posted 
https://forums.unraid.net/topic/151360-how-does-one-add-an-app-to-the-community-addons/
so hopefully someone replies on there i havent tried discord yet

davidzwa, 2024-01-17:
https://github.com/selfhosters/unRAID-CA-templates/blob/master/templates/adminer.xml

It seems we need a xml file to be present. This is not too complex.

davidzwa, 2024-01-19:
I've added the XML, but I've not been able to verify it as I do not have Unraid installed.
https://github.com/fdm-monster/fdm-monster-unraid

We must be able to test the container before this app is requested to go live. I do not have a physical server available right now. I do have an old PC, but it is surely not up for the task. 

Any advice on whether Unraid can be virtualized, run as cloud server, etc?

davidzwa, 2024-01-23:
@comet424 it would be great to hear back from you. If not I will have to schedule setting up an Unraid server myself, which might take a while as I do not have the resources set up at this moment.

comet424, 2024-01-24:
@davidzwa sorry havent been on the computer

ya i guess unraid can be run virtualzied...   unraid its a server that runs  virtual machines itself... so it runs off a usb key 

but for your program showing up on the Community Apps  it hasnt shown up there least not yet  when i search for 
fdm monster   unless its under a different name

davidzwa, 2024-01-24:
I have only made the repository and added the xml file. I hav not reached out to an Unraid app moderator.

Since the XML has not been tested yet, it does not seem wise to publish it without testing.

Could you point me to a tutorial on how to set up such a virtual machine with USB forwarding? The ones I found assume I have the hardware ready for it

comet424, 2024-01-24:
gotcha  how do i install your repository with that wget command?

as for unraid.. if you   if you got a old computer kicking around or a laptop   that your not using....  
download  the usb installer  from unraid.net  install the latest unraid to the usb.. and it will default to tower...  and then boot off the usb... and then ur done.. unraid  is now running off that computer...  youll get a 30 day trial off that usb with 2 extra 15 day trials after that...  as unraid will run off any hardware pretty much... and i can help you from there...  
and youll get a ip address  at the login  ull use that ip address on another computer   in your browser to access it... oh and u need a spare hard drive  just for the array   but thats all..  but i can test  i just dnt know how to install it

but i can try installing your app i just dont know how to use  the command lines

comet424, 2024-01-24:
@davidzwa  and this might help too this tutorial 
https://www.youtube.com/watch?v=huCE4jtXOjQ

davidzwa, 2024-03-31:
Im closing this issue - and with great pleasure I must add

closed 2024-03-31 by davidzwa