fdm-monster/fdm-monster #2889: Option to add camera stream to floors page

ZombiesLoveMe, 2024-02-12:
**Describe the solution you'd like**
Addition of live camera stream into the floors page for each printer container, this is to allow quick viewing of printer state

davidzwa, 2024-02-12:
[image]
I could definitely see this happening on this spot of the UI.

Otherwise an 👁️ icon on a small printer 'tile' and a full fledged camera view for large printer 'tile'.
[image]
_large tile mode_
[image]
_small tile mode_

davidzwa, 2024-02-12:
@all-contributors add @ZombiesLoveMe for idea

davidzwa, 2025-02-22:
Due to a milestone which has received other features than planned we are heading into bugfix milestones (like 1.8.3). Unfortunately this feature has to be scoped down during refinement which does not fit in 1.8.3 but rather a minor release (f.e. 1.9 or 1.10). For that to happen this has to be postponed and put back on the backlog.