fdm-monster/fdm-monster #2891: Basic printer controls (preheating, filament change and movement)

ZombiesLoveMe, 2024-02-12:
**Is your feature request related to a problem? Please describe.**
Problem is that i have to leave FDM monster to control the printer adding an additional step if this was brought in house it would mean less work from my end

**Describe the solution you'd like**
addition of basic printer controls Eg movements, extrude preheat etc

davidzwa, 2024-02-12:
Very nice!

My list of prioritized items would be:
- printer controls (movement + speed)
- printer controls (extruder)
- temperature view
- preheat
- filament retract (preconfigured distance)
- leads to: filament change (with or without plugin)

I will have to split these out as the work progresses.

davidzwa, 2025-02-22:
This feature request is great. The problem however arises in the fact that the current control dialog that is present in FDMM since client 1.5.4 and server 1.6.1 does not have a good user experience (UX). For example rapidly clicking a home and movement button causes the API calls to OctoPrint to spill over and sometimes fail slowly, there's no queueing and cancelation nor does the UI have any action queueing of sorts. In other words: the control dialog does not work smoothly and that would mean any further expansion could result in poor UX.

We have to refine this issue further and thus it's best to move it back to the backlog with `refinement` label so we can treat this during refinement.