fdm-monster/fdm-monster #3047: Add maintenance items to printers.

Mikec78660, 2024-03-31:
A field in the box that opens up when you select a printer from the grid, maybe under put printer in maintenance mode, that would pop up a new window. The new window would allow for inputting service items on the printer. Allow a user configurable drop down list for each printer of maintenance items. So you select a maintenance item, say replace nozzle and select the date. Maybe even allow for items to be visible in the printer grid all the time. So it would have nozzle and hot end and whatever else you wanted displayed with their age, calculated from the last replacement. Maybe even allow an option when adding a new printer to FDM Monster to ask if the printer is new and if so set all the components to that days date. So if a print failed it would be quick to see that the nozzle was last replaced 8 months ago and maybe that would be a place to start.

Right now we are using an excel sheet to keep track of all printers and maintenance. It would be great if that could be done inside FDM Monster. Maybe even log the person who added the item by their login, or let you select the person who did the maintenance from a drop down of FDM Monster users.

P.S. I really love this software!

davidzwa, 2024-03-31:
It's awesome to hear from you! This feedback is great. 

Let me summarize your words and ask some questions in return. I have built this feature before for another project, and depending on your reply we can tweak what I built before to better suit FDM Monster being an unopinionated project.
_Small note_: this reply may seem daunting. Do not worry, I have a strategy in mind and not everything has to be built all at once.

### If I summarize your suggestion for this feature it involves:
- the maintenance dialog
- the printer tile
- the create printer dialog (for immediate consistency)

### What I am missing in the topics is:
- a repair-log/maintenance admin page + export CSV / JSON / YAML (whatever is preferred)

### About the maintenance dialog:
- you would like to see the maintenance dialog extended with at least a multi-select dropdown for parts to be replaced
- you would suggest to make this list of parts configurable from settings as different approaches will exist (and I agree)
- you would like to see an assignee dropdown, also based on settings
- you would like to have a date selector, preferably with the current date already selected 
  - this suggests you want to backtrack printers that broke down before - am I correct here?
  - we need some discussion: what if someone wants to add a maintenance log without setting the printer to in maintenance? My idea would be to use the repair log page to "fill in a previous repair attempt" so the printer can stay enabled and the repair log is more consistent. Alternatively we can decide to add a checkbox called "disable printer" which is checked by default to the maintenance dialog.

### About the create/update printer settings dialog (both create and update have the same dialog😉):
- you would like to set maintenance from these settings immediately (as well)

### About the printer tile:
- you would like to show the last date of a "part" since their last replacement, mainly for specific parts of choice
  - I think this is a feature we should only enable for "large tile mode" as the information wont fit on small tiles
  - we need to go back on forth on this, I dont have a simple UI design in my mind yet
  
For now I propose that I will start on extending the maintenance dialog, the parts settings, the assignee and implementing the repair log page. 
Would be great to hear back from you.

Mikec78660, 2024-03-31:
Wow, yes. That's exactly it. And yes the export CSV would be great to make it easier to calculate things like repair costs and things that might be beyond the scope of FDM Monster. While cost of parts used and time spent would be cool to track in FDM Monster it might be beyond it's original intent. If it's not then you could go all the way and calculate things like filament used and total print time to each part. So not only would you see that it has been 8 months since you replaced the x axis bearing wheels but you could see that the machine has 3,000 print hours since you replaced them.

maybe let the community create a lit of parts that would populate in the maintenance log depending on the printer model. So you could download a json file or something for different printer models to save people time not having to build out a list from scratch if someone else has already done it.

If you wanted to really go crazy inventory management of parts could be done. It would require a database of inventory of spare parts to be maintained and when a replacement is logged it would subtract one from the inventory. But that would require discipline of users to keep up with a and probably a limited number of the users would utilize this. The shops disciplined and large enough to use such a feature probably already have their own system in place. But it would be cool. I guess it depends on the end goal of FDM Monster. Is it designed to monitor and submit print jobs or is it trying to be a total print farm management system.

davidzwa, 2024-03-31:
You strike some good points: whether FDM Monster is a full-fledged farm platform or a multi-printer job tracker depends on the bandwidth of shipping features and fixing bugs.

So far I've found it challenging to involve other developers consistently. Maybe we should shift the attention there and see how we can make the project thrive by attracting other developers and/or bringing out in the open more.
My gut feeling says that either the project is not well-known enough or the codebase could be more attractive to work on.
I have good faith that both challenges can be solved with commitment and time.

Mikec78660, 2024-03-31:
Well there is for sure a need for a project like yours. I have watched countless youtube videos when debating if and how to set up a printer farm. So many people with dozens of printers are just using sd cards to move prints to printers. They have to physically monitor each printer to know it's status and have no idea what their cost of ownership is for their printers.

I started with octoprint when I got my first 3d printer about 5 years ago so when I wanted to add more obviously octofarm popped up but it doesn't seem to be developed anymore. So I found posts that directed me to your project and I think it is great.  Being able to run 4 printers from each pi and controlling them all from your project seems light years ahead of running around from printer to printer to see when jobs might finish. And adding a cam to detect failures and stop them quickly. I don't know how people run farms without software like yours.

But one more quick feature request. Adding a link to the camera feed in the grid would be great. Right now you add a feed that seems totally separate from the printer. Would be nice if you added the feed when you add or edit a printer so you could link them together with the same name and everything (the printer info and video feed I mean).

davidzwa, 2024-03-31:
I've made a separate issue for that https://github.com/fdm-monster/fdm-monster/issues/3048
Thanks for bringing it to my attention again.

davidzwa, 2024-03-31:
@all-contributors add @Mikec78660 for idea