fdm-monster/fdm-monster #3110: Change button colors

mkevenaar, 2024-04-15:
**Describe the bug**
The current buttons are red, it screams, it suggests things are broken/bad/etc.

**To Reproduce**
NVT

**Screenshots**
[image]

Suggestion to change:

(changed CSS class primary to gray)
[image]

**Browser (please complete the following information):**

**Additional context**

davidzwa, 2024-04-18:
Associated issue on client side (https://github.com/fdm-monster/fdm-monster-client/issues/86). 

Let's keep that one in mind when tackling this issue.

davidzwa, 2024-12-14:
This issue is less work now. 

- Make (grid) settings self-saving
- Remove any primary color on buttons which need to be kept (autosave not possible due to complexity)

closed 2024-12-17 by davidzwa