fdm-monster/fdm-monster #3169: Getting errors when try to view the printers details

shandshellin, 2024-04-26:
I can create the printers and add cameras but no controls and if i view details i keep getting error 500. 

AN ERROR OCCURRED
Request failed with status code 500 - URL /api/printer-files/1/?recursive=false

Here are the logs

[fdm-monster-2024-04-26.log](https://github.com/fdm-monster/fdm-monster/files/15134669/fdm-monster-2024-04-26.log)
[pm2.error.log](https://github.com/fdm-monster/fdm-monster/files/15134670/pm2.error.log)
[pm2.log](https://github.com/fdm-monster/fdm-monster/files/15134671/pm2.log)

davidzwa, 2024-04-26:
I do see in your logs that you are getting errors while files are being stored to database. This action is called when the side panel is opened.
One other error I see is "self signed certificate". Please check if this is a problem as well.

In order to help you better I need details that the log will not show. Therefore, could you enable diagnostics?
https://docs.fdm-monster.net/docs/troubleshooting#sentry

shandshellin, 2024-04-26:
.log files are added to the first post, do you not see them?

shandshellin, 2024-04-26:
sentry enabled

davidzwa, 2024-04-27:
I received the error in Sentry. Its ok to disable diagnostics again.

davidzwa, 2024-04-27:
QueryFailedError: SqliteError: NOT NULL constraint failed: printer_file.hash

I will this asap.

closed 2024-04-27 by davidzwa

shandshellin, 2024-04-27:
I can see the files now, thanks. The issue now is the printer controls. If i refresh the page i can open the controls once for one printer and then they are just blank, then refresh the page and i can open one of the controls. Thoughts?

shandshellin, 2024-04-27:
i enabled diag again.