fdm-monster/fdm-monster #3180: add a local gcode storage when doing print uploads

iandouglas, 2024-05-01:
Per Discord discussion:

The feature request is to add a local gcode storage which can be referenced when doing multi-select print uploads. When a file is selected which the printer does not have (or which is newer from the local source), optionally purge the file and upload it.

Requirement #1: upload progress bar in the webpage should be improved.

Requirement #2: improve/test 400MB+ uploads for FDM Monster API, ensure any upload limits are configurable in settings and documented.

davidzwa, 2024-05-01:
@all-contributors add @iandouglas for idea

davidzwa, 2024-05-01:
Thanks @iandouglas. Once this issue receives a milestone, the feature (epic) will be released with that version with high probability. Note this issue will be closed when the feature has been merged. The released moment might still be a while after, depending on the other work items to be completed first.

#### Addition technical notes to self for later:
When the standardized "printer API" has been implemented, this epic can be tackled. The Printer API should be extended to include an optional setting or check when uploading a file, or when selecting a locally stored GCode (new feature in this feature request issue). This check performs the following steps:
- FDM Monster performs a file query to the printer API and receives a file's path, size and hash
- FDMM receives the upload or checks the local storage. It now performs a check whether a hash calculation is required. If the file name is the same, start a hash calculation procedure
- FDM Monster compares the hash (if so required) and if hash mismatches, performs a printer API file delete.
- Once file delete is completely (successfully), the new file is uploaded through the Printer API

We should keep in mind that this feature might have impact on FDM Monster server CPU usage:
- Pay close attention to CPU usage and check if a NodeJS worker thread is needed.