fdm-monster/fdm-monster #3335: Use of deprecated term Emergency Stop which does not meet legal requirements

InsanityAutomation, 2024-06-21:
**Describe the bug**
The term Emergency Stop is utilized across the UI in many places. This is a legally protected term in most countries with specific functional requirements behind it. This creates legal liability in the event it fails to prevent an incident for both the user and the project team. A nomenclature change avoids the false representation liability issue. Quick stop, Abort, Full Shutdown.. These are common terms used for the function today.

**To Reproduce**
Steps to reproduce the behavior:
>
NA
> 
**Expected behavior**
>
NA
**Screenshots**
If applicable, add screenshots to help explain your problem.
>

**Browser (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [1.1.0, etc.]

**Additional context**
Add any other context about the problem here.

https://marlinfw.org/docs/gcode/M112.html
https://github.com/MarlinFirmware/Marlin/pull/15065

>

davidzwa, 2024-06-22:
Thanks for supplying the informatie and options in regard to the alternatives. I've selected "Quick Stop" as I find "Full Shutdown" (Marlin's page) confusing in a context where multiple devices are operational for FDM Monster to work. It suggests OctoPrint as well as the printer might need to be booted up after by manual intervention.
The term "Quick Stop" conveys the reality: it is much faster than cancelling in any way or form.

davidzwa, 2024-06-22:
@all-contributors add @InsanityAutomation for idea bug

closed 2024-06-22 by davidzwa