fdm-monster/fdm-monster #3346: Add additional authentication & SSO methods

LyfeOnEdge, 2024-06-25:
FDM Monster would likely see more adoption if it supported a wider range of authentication methods.

Here are a few options, the Proxy + LDAP (SSO)  option would be awesome but if you were to pick the easiest / fastest it would probably be the Proxy-Auth that offloads authentication to a proxy.

**LDAP tie-in (Authentication):**
Many users will have an existing ldap/AD system, an integration with ldap would make user management easier for larger groups.
- privileged: `admins` or another group like `printeradmins`
- non-privileged users: `everybody` or another group like `printerusers`
Apache Guacamole has a good example of what the config would look like:
https://guacamole.incubator.apache.org/doc/gug/ldap-auth.html

**Proxy-Auth (SSO):**
Many users will put FDM Monster behind a proxy auth system.
Authelia and Authentik are popular examples of docker containers where the proxy passes the username or email and groups of a user to FDM Monster. If the user doesn't exist it is created in the local database.
Grafana has a good example of what the config would look like:
https://grafana.com/docs/grafana/latest/setup-grafana/configure-security/configure-authentication/auth-proxy/

**Proxy + LDAP (SSO) (more secure than pure Proxy-Auth):**
This is a combination of the above two options, the proxy sends the header to FDM Monster and FDM Monster uses this info to check ldap to confirm the user is permitted.

Thank you for considering this, this project is awesome and I've enjoyed watching it grow

davidzwa, 2024-10-26:
@all-contributors add @LyfeOnEdge for idea

AndrewSpangler, 2024-11-12:
After talking with @davidzwa, a bring-your-own auth using a trusted proxy seems like the best approach to adding scalable, easily integrated enterprise-style RBAC to FDM-Monster. 

**Role-based access control and user auth via trusted proxy headers**

Goals:
    - Add abstract user-based RBAC to FDM-Monster by leveraging proxy-auth for better scaling and enterprise-style integration
    - System should be flexible, allowing customization of which headers are used for authentication purposes, and should be able to accept an email, id, or username. It should also allow internal roles to be mapped to external roles ("groups"), to prevent conflicts with existing systems.
    - System should be secure, requiring a "trusted proxy" at a static IP that will send proxied requests to FDM-Monster
    - System should be fault-tolerant, at a minimum maintaining the ability for an admin to log in locally if the external auth system is disabled. Optimally, proxy users will be asked for a local-login password the first time that could also be optionally used to provide a second layer of security if desired. 
    - Long-term goals should include documentation on configuring FDM-Monster with some of the common open-source proxy/auth solutions like Authentik/Nginx Proxy Manager, Authelia/Traefik, etc. 

Details:
    - Reverse-proxies can add headers to the servers they proxy requests to, this can give a server extra context about the request.
    - Proxies may implement a concept known as "middleware", which can do things to a user's request such as redirect a user to a login screen before allowing them to access an endpoint.
    - These proxy/middleware systems are commonplace in enterprise environments and scale much better than many independent locally hosted databases.

Specifics:
    - Proxy-Auth systems may use many combinations of a variety of headers but they usually boil down into two groups:
        - User Identifier:
            This is usually the user's name, email, uid, or employee ID.
            Common headers include:
            - remote-user (Popular w/ Authelia)
            - remote-email (Popular w/ Authelia)
            - remote-id (Popular w/ Authelia)
            - X-Forwarded-User (Popular w/ Nginx Proxy Manger)
        - Group Membership:
            This is a list of groups the user might be part of.
            Common headers include:
            - remote-groups (Popular w/ Authelia)
            - X-Forwarded-Groups (Popular w/ Nginx Proxy Manger)
    - Overhead look of proxy-auth:
        -> User accesses server via proxy
        -> User is redirected to proxy middleware login screen to log in
        -> User is redirected to server
        -> Headers are sent by the proxy to the server with the request
        -> Server sees proxy headers and verifies proxy is a trusted IP or responds with a 403
        -> If the server doesn't see user in local user cache DB, creates a user
        -> Request local login / optional secondary login password from user (so if somebody does manage to breach your local network and impersonate the proxy there's still a login etc)
        -> Server uses header to check user roles, if they don't match local user cache DB, update user roles in cache db
        -> Server checks user permissions to ensure they have access to the endpoint the are requesting
        -> Serve content if user has permission or respond with 403 code if they are forbidden