fdm-monster/fdm-monster #3408: Enhancing the FDM-Monster database

MoonsterAlpha, 2024-07-17:
It would be interesting to obtain, like OctoPrint's ‘PrintJobHistory’ plug-in, a similarly detailed database of each print job carried out for instances connected to FDM-Monster.
This would ultimately be a global PrintJobHistory for all instances, with the same extraction functionalities per period.

This could be done using the current Mongodb database managed by FDM-Monster. By asking the user to install the PrintJobHistory plugin on each of their OctoPrint instances and retrieving the contents of the local databases. This could also be done by taking inspiration from the existing plugin and extending it to all instances, which would be a feature specific to FDM-Monster.

davidzwa, 2024-11-28:
@MoonsterAlpha I have other plans around history. My proposal would be to not manage a third-party plugin, but instead track history through FDM Monster itself by use of GCode analysis. This is in the works for #1906 and a similar approach works for extracting estimated time, filament usage details (weight, volume, density, diameter).

We could think about importing history (backfilling) from such a plugin, excel or custom scripts using separate tooling outside FDM Monster itself.

Would this be in line with your idea? What was your reasoning for bringing up the plugin?

davidzwa, 2025-02-22:
This issue is duplicate of #2845 given the direction we want to go. Closing for that reason.

closed 2025-02-22 by davidzwa