fdm-monster/fdm-monster #3454: Bug when view details on printer

V4LVE, 2024-08-06:
**Describe the bug**
When entering the details of a printer from the floor. A 500 error occours

**To Reproduce**
Steps to reproduce the behavior:
> Enter details of a printer from the floor.
> 
**Expected behavior**
> No fault codes

**Screenshots**
If applicable, add screenshots to help explain your problem.
> 
[image]

**Browser (please complete the following information):**
 - OS: Windows (Docker)
 - Browser: chrome
 - Version: FDMM: Latest. OctoPrint: Latest

davidzwa, 2024-10-26:
Hi @V4LVE can you tell me if this problem occurs regularly?

Is this OctoPrint maybe showing trouble loading its webpage or maybe its slow?
Have you tried increasing the timeout setting of FDM Monster?

And finally could you let me know if the problem persists in 1.6.4?

Thanks :)

davidzwa, 2024-11-14:
@V4LVE eager to hear your reply.

davidzwa, 2024-12-14:
Do to lack of response I am closing this issue. FDM Monster has had a lot of work and without further information it is hard for me to say if this issue is still present.

closed 2024-12-14 by davidzwa