fdm-monster/fdm-monster #3464: Unable to change password or modify user roles

callanova, 2024-08-09:
**Describe the bug**
When navigating to http://localhost:4000/settings/account, it states:
```
Login is currently disabled. To adjust your username and password, please enable that setting at the Server Protection settings page. Then log in and visit this page.
```
Within http://localhost:4000/settings/server-protection, Login Required > Require Login checkbox is enabled.
If a user registers a new account, I can verify the account via http://localhost:4000/settings/user-management and set it as a owner. However, the new account still has the GUEST role. When I login as that new account, I get "Permission Denied" message with the following (The problematic API path varies):
```
You do not have permission to access a specific resource on this page.
Page: Home
Usable roles: OPERATOR, ADMIN
Problematic API path: /api/printer
Error: Authorization failed
```
I'm unable to modify the roles to remove GUEST or add OPERATOR.

**To Reproduce**
Steps to reproduce the behavior:
> Create a docker-compose.yaml file with the following contents:
```
services:
  mongodb:
    image: mongo:7
    container_name: mongodb
    environment:
      # Authenticated variant, comment out for non-authenticated variant:
      - MONGO_INITDB_ROOT_USERNAME=root
      - MONGO_INITDB_ROOT_PASSWORD=fdm-root
    ports:
      - "28017:27017"
    volumes:
      - ./mongodb-data:/data/db
      - ./mongoconfig:/data/configdb
    restart: always

  fdm-monster:
    container_name: fdm-monster
    image: fdmmonster/fdm-monster:1.6.3
    restart: always
    ports:
      - "4000:4000"
    environment:
      # Authenticated variant:
      - MONGO=mongodb://root:fdm-root@mongodb:27017/fdm-monster?authSource=admin
      - SERVER_PORT=4000
    volumes:
      - ./fdm-monster/media/:/app/media

```
> Run `docker-compose up -d`
> Navigate to http://localhost:4000/ in a non private browser
> Go through the initial setup and enable Registration
> Log in with the account that was created
> Open a private browser
> Go through the steps to register a new account
> Navigate to http://localhost:4000/settings/user-management that is logged in with initial account
> Verify the new user
> Try to modify the user's Roles
> Try to modify the password for either account at http://localhost:4000/settings/account
**Expected behavior**
> Able to modify user Roles
> Able to change password

**Screenshots**
If applicable, add screenshots to help explain your problem.
> 
[image]
> 
[image]
> 
[image]
> 
[image]

**Browser (please complete the following information):**
 - OS: Windows 11
 - Browser: Edge
 - Version:
 ```
Current versions in use:
Your server's version is: 1.6.3
Your client's version is: 1.5.14
```

**Additional context**
Add any other context about the problem here.

>

davidzwa, 2024-11-11:
Hi I will introduce a new UI for setting the user's roles upon verification. I agree that adding the GUEST role makes no sense, if there is nothing the user can do with it. Therefore it is most definitely a bug.

You can expect this change for release 1.7.3 as I do see that this can hamper use of FDM Monster with multiple accounts.

davidzwa, 2024-11-11:
@all-contributors add @callanova for bug

davidzwa, 2024-11-11:
I've split off this issue into two bugs. This one covers the roles, the other one is about not being able to change your password https://github.com/fdm-monster/fdm-monster/issues/3764

PR #3759 (2024-11-11): Fix/set roles api
Fixes #3464 by adding the ability to set user roles. This is achieved by adding a new endpoint.