fdm-monster/fdm-monster #3726: [Klipper]: Printer:Error

johnbxl1970, 2024-11-04:
### Checkboxes

- [X] I agree to follow our [Code of Conduct](https://github.com/fdm-monster/fdm-monster/blob/develop/CODE_OF_CONDUCT.md).
- [X] I have verified no [other issues](https://github.com/fdm-monster/fdm-monster/issues) exist related to my request.

### What You Are Seeing?

When clicking on a printer under the Printer section for more details. The section "Printer" under printer details has the text "Error" and never changes.
[image]

### What is Expected?

I don't know as I don't have printer with OctoPrint to compare

### How Did You Get This To Happen?

Add printer with "+ Create Printer" then select "Klipper" and add printer connection information.

### System Details

* Operating System and version: Ubuntu 24.04
* Browser and version: Chrome Version 130.0.6723.92
* FDM Monster Installation Method: Docker
* FDM Monster Server version: 17.0
* FDM Monster Client version:  16.5
* Moonraker version: v0.9.3-1-g4e00a076
* Klipper version: 09faed31-dirty

### Additional Context

Same for Creality Ender3 v3 and Creality Ender3 v3 KE

davidzwa, 2024-11-04:
At this point in time my intuition says that this "Error" message is a dummy generated in case there is no display message object.
This is interesting.

@johnbxl1970 are you able to provide me with the available objects that your moonraker is configured with / capable of sending?

Since I'd need a bit more information, I will push this away from `1.7.1` until more information is available.

johnbxl1970, 2024-11-04:
Hi David, here is the output from the printer. Hoep this will help :)
`{"result": {"objects": ["webhooks", "configfile", "mcu", "mcu nozzle_mcu", "mcu leveling_mcu", "mcu rpi", "virtual_pins", "gcode_macro xyz_ready", "gcode_macro _IF_HOME_Z", "gcode_macro _IF_MOVE_XY", "gcode_macro _HOME_X", "gcode_macro _HOME_Y", "gcode_macro _HOME_Z", "gcode_macro PRINTER_PARAM", "gcode_macro STRUCTURE_PARAM", "gcode_macro AUTOTUNE_SHAPERS", "gcode_macro LOAD_MATERIAL_CLOSE_FAN2", "gcode_macro LOAD_MATERIAL_RESTORE_FAN2", "gcode_macro SET_E_MIN_CURRENT", "gcode_macro RESTORE_E_CURRENT", "gcode_macro LOAD_MATERIAL", "gcode_macro QUIT_MATERIAL", "gcode_macro Qmode", "gcode_macro Qmode_exit", "gcode_macro M204", "gcode_macro M205", "gcode_macro M106", "gcode_macro M107", "gcode_macro M900", "gcode_macro WAIT_TEMP_START", "gcode_macro WAIT_TEMP_END", "gcode_macro PRINT_CALIBRATION", "gcode_macro FIRST_FLOOR_PAUSE_POSITION", "gcode_macro ACCURATE_G28", "gcode_macro Z_SENSORLESS_SET", "gcode_macro Z_SENSORLESS_CLEAR", "gcode_macro PRINT_PREPARED", "gcode_macro PRINT_PREPARE_CLEAR", "gcode_macro END_PRINT_POINT_WITHOUT_LIFTING", "gcode_macro END_PRINT_POINT", "gcode_macro END_SELF_TEST_POINT", "gcode_macro END_PRINT", "gcode_macro END_SELF_TEST", "gcode_macro FIRST_FLOOR_PAUSE", "gcode_macro FIRST_FLOOR_RESUME", "gcode_macro PAUSE", "gcode_macro INPUTSHAPER", "gcode_macro BEDPID", "gcode_macro TUNOFFINPUTSHAPER", "gcode_macro RESUME", "gcode_macro CANCEL_PRINT", "gcode_macro G29", "gcode_move", "print_stats", "fan_feedback", "custom_macro", "heaters", "output_pin ADAPTIVE_BED_MESH", "output_pin FULL_BED_MESH", "output_pin ADAPTIVE_PURGE_LINE", "gcode_macro _ADAPTIVE_BED_MESH_ON", "gcode_macro _FULL_BED_MESH_ON", "gcode_macro _BED_MESH_OFF", "gcode_macro _ADAPTIVE_PURGE_LINE_ON", "gcode_macro _ADAPTIVE_PURGE_LINE_OFF", "gcode_macro KAMP_BED_MESH_SETTINGS", "gcode_macro KAMP_PURGE_LINE_SETTINGS", "gcode_macro START_PRINT", "gcode_macro BED_MESH_CALIBRATE", "gcode_macro _LINE_PURGE", "gcode_macro _SMART_PARK", "gcode_macro DEFINE_OBJECT", "gcode_macro START_CURRENT_OBJECT", "gcode_macro END_CURRENT_OBJECT", "gcode_macro LIST_OBJECTS", "gcode_macro LIST_EXCLUDED_OBJECTS", "gcode_macro REMOVE_ALL_EXCLUDED", "gcode_macro _KAMP_Settings", "gcode_macro _GIT_STOP", "gcode_macro _GIT_PAUSE", "gcode_macro _GIT_RESUME", "gcode_macro GIT_BACKUP_STOP", "gcode_macro GIT_BACKUP_PAUSE", "gcode_macro GIT_BACKUP_RESUME", "gcode_macro CAM_SETTINGS", "gcode_macro CAM_BRIGHTNESS", "gcode_macro CAM_CONTRAST", "gcode_macro CAM_SATURATION", "gcode_macro CAM_HUE", "gcode_macro CAM_WHITE_BALANCE_TEMPERATURE_AUTO", "gcode_macro GET_TIMELAPSE_SETUP", "gcode_macro _SET_TIMELAPSE_SETUP", "gcode_macro TIMELAPSE_TAKE_FRAME", "gcode_macro _TIMELAPSE_NEW_FRAME", "gcode_macro HYPERLAPSE", "gcode_macro TIMELAPSE_RENDER", "gcode_macro TEST_STREAM_DELAY", "probe", "pause_resume", "filament_switch_sensor filament_sensor", "idle_timeout", "virtual_sdcard", "temperature_fan soc_fan", "tmc2209 stepper_x", "tmc2209 stepper_y", "tmc2209 stepper_z", "tmc2209 extruder", "heater_bed", "heater_fan my_nozzle_fan", "output_pin fan0", "output_pin fan2", "output_pin fan1", "bed_mesh", "display_status", "exclude_object", "motion_report", "query_endstops", "system_stats", "manual_probe", "toolhead", "extruder"]}}`

davidzwa, 2024-11-16:
Hi, Ive found another user that has the same text appearing. We have good reason to believe it is not a problem that will cause the ability to operate the printer to be hampered.

Regardless, I intend to dig into this issue soon!