fdm-monster/fdm-monster #3747: [Klipper]: Printer information displaying mainsail

johnbxl1970, 2024-11-07:
### Checkboxes

- [X] I agree to follow our [Code of Conduct](https://github.com/fdm-monster/fdm-monster/blob/develop/CODE_OF_CONDUCT.md).
- [X] I have verified no [other issues](https://github.com/fdm-monster/fdm-monster/issues) exist related to my request.

### What You Are Seeing?

When clicking on more information for a printer from the dashboard for a printer with fluid, it displays "Mainsail" instead and the url defaults to http:xxx.xxx.xxx.xxx:8080.
[image]

### What is Expected?

I would expect that it detects it is fluid from the printer configuration or that there is a switch to select either using Mainsail or Fluidd.

### How Did You Get This To Happen?

From the Dashboard, click "More information" on the printer, then it will be cvisible in the pop-out on the left.

### System Details

Operating System and version: Ubuntu 24.04
Browser and version: Chrome Version 130.0.6723.92
FDM Monster Installation Method: Docker
FDM Monster Server version: 17.0
FDM Monster Client version: 16.5
Moonraker version: v0.9.3-1-g4e00a076
Klipper version: 09faed31-dirty

mkevenaar, 2024-11-09:
## Suggested fix

Identify if the front-end used is either mainsail or fluid.
If identification is impossible, give the user a drop-down with either option.

johnbxl1970, 2024-11-09:
A drop down list in the printer confi when connecting the printer is absolutely fine

davidzwa, 2024-11-11:
I just did some investigation. It is not straight-forward to auto detect the UI used by asking Moonraker. Therefore, we will need to resort to a dropdown or input field with radio button. Some URLs need to be changed if users changed default ports, or have domain names/ssl, proxies, etc.

Since FDMM now needs to store printer service related data and this causes need for a migration, I cannot assign this to a bugfix release. Moving to 1.8.0 instead.

davidzwa, 2025-02-22:
Sadly this has not been tackled in 1.8.0. Moving to 1.10 for klipper related features.

thatguy-jaysenodell, 2025-03-29:
Some unsolicited feedback: I host mainsail as a standalone docker on a separate host, mainsail.jodell.net,  from all the printer hosts. Using config.json I have the hosts predefined in mainsail. EX
`{
  "instancesDB": "json",
  "instances": [
     { "hostname": "e5a.jodell.net", "port": 7125 },
     { "hostname": "e5b.jodell.net", "port": 7125 }
   ]
}`

The ability to provide a separate UI host/port should be provided for cases were hardware optimization is the focus. This philosophy is already evident in the camera configuration options (allowing for non-printer hosted cameras).