fdm-monster/fdm-monster #3880: [Bug]: Thumbnail images are duplicated across printer floors, don't display immediately.

Tbwan, 2024-12-04:
### Checkboxes

- [X] I agree to follow our [Code of Conduct](https://github.com/fdm-monster/fdm-monster/blob/develop/CODE_OF_CONDUCT.md).
- [X] I have verified no [other issues](https://github.com/fdm-monster/fdm-monster/issues) exist related to my request.

### What You Are Seeing?

Issue 1: When activating the experimental thumbnail preview feature, I noticed that the thumbnails for currently printing files don't always show up immediately. Refreshing the webpage doesn't seem to fix it, but they do display after navigating to another menu on the left-hand side (i.e. printers, cameras, or statistics) and back again.

Issue 2: I also noticed that when multiple printer floors are in use, thumbnail images are duplicated on the same grid location in every printer floor. I wonder if this may be part of the cause of Issue 1, due to conflicts of thumbnail display attempts across printer floors.

### What is Expected?

1 - Thumbnail images should display immediately instead of needing the use to navigate to another menu and back again.
2 - They should also be isolated to just the relevant printer, and not duplicated across all printer floors. They should also

### How Did You Get This To Happen?

1. Enabled the "Experimental Thumbnail Support" feature
2. Sent a file to one of the printers
3. Navigated to a different left-hand menu page, then back again (this seems to cause the thumbnails to show up)
4. Navigated across different printer floors (to discover that the thumbnail images are duplicated.)

### System Details

* Operating System and version: Windows 10 Enterprise 10.0.19045 Build 19045
* Browser and version: Firefox 133.0 64bit
* FDM Monster Installation Method: Docker
* FDM Monster Server version: 1.8.0
* FDM Monster Client version:  1.8.2
* Middleware: Moonraker
* Middleware version: Moonraker v0.9.3-3-gccfe32f2

davidzwa, 2024-12-05:
@all-contributors add @Tbwan for bug

davidzwa, 2024-12-07:
Issue 2 is tracked in #3892. Changing title to "Thumbnails are only reactively, not on page load"

closed 2024-12-07 by davidzwa