fdm-monster/fdm-monster #3921: [Feature]: Provide a kubernetes installation guide.

nuvious, 2024-12-14:
### Checklist

- [X] I agree to follow our [Code of Conduct](https://github.com/fdm-monster/fdm-monster/blob/develop/CODE_OF_CONDUCT.md).
- [X] I have verified no [other issues](https://github.com/fdm-monster/fdm-monster/issues) exist related to my request.

### Is Your Feature Request Related To A Problem? Please describe.

No, this would just be template configuration files to make available to the community for those running services through kubernetes.

### Describe The Solution. Why is it needed?

Provide a configuration template that can be customized and deployed with the kustomize declarative management functionality of kubernetes:

https://kubernetes.io/docs/tasks/manage-kubernetes-objects/kustomization/#customizing

### Additional Context

I run a kubernetes cluster in my home lab and try to keep all my services running in that cluster, mainly because I can use network policies to harden my internal network. Other people in the home lab community would benefit agnostic to my needs if they're using kubernetes because it enables deployment without spinning up a dedicated VM or running the service on a machine running other services concurrently.

NOTE: Pull request pending. Just getting an issue number so I can reference it in my pull request and branch.

davidzwa, 2024-12-15:
@all-contributors add @nuvious for ideas, platform, infra

davidzwa, 2025-02-22:
This issue was brought on the roadmap a bit too quick. It has been marked ready for refinement and will for now be moved back to backlog so we dont forget.