fdm-monster/fdm-monster #4164: [Klipper]: Can not start batch job from upload.

thatguy-jaysenodell, 2025-03-24:
### Checkboxes

- [x] I agree to follow our [Code of Conduct](https://github.com/fdm-monster/fdm-monster/blob/develop/CODE_OF_CONDUCT.md).
- [x] I have verified no [other issues](https://github.com/fdm-monster/fdm-monster/issues) exist related to my request.

### What You Are Seeing?

Current config:
- Docker image hosted
- 2 moonraker based systems (kA and kB) confirmed working via individual printer file upload/print start/monitoring and controls.

Use case:
A gcode file has been created from <slicer> and should be executed on a set of printers. A FDMM batch job is created to orchastrate the upload, execution and monitoring of the print process. 

Result:
- Files are uploaded
- Files are not queued for immeadiate or future printing. 
- No patch job history is created. 
- If the files are manually executed then stopped, then a batch job will be visiable for scheduling. 

### What is Expected?

- The selected file would be uploaded to selected printers and placed into the print queue. 
- Based on Octoprint behavior, if there are no other prints in queue, print should begin immediately. 
- The batch job would be registered in the job history for future reference/execution.

### How Did You Get This To Happen?

- Llogin to FDMM and select kA and kB.
- In the batch job header select gcode file (or drag/drop).
- Click upload file
- Wait

### System Details

* Operating System and version: FDMM Docker image (running on Linux/Debian bookworm), browser is MacOS 15.3.2 Intel
* Browser and version: Safari 18.3.1
* FDM Monster Installation Method: Docker (latest)
* FDM Monster Server version:  1.8.3
* FDM Monster Client version: 1.9.0
* Moonraker version: 0.9.3
* Klipper version: 0.12

### Additional Context

I like the idea of uploading files and not having them immediately printed but there needs to be a method of selecting when that happens. That or the ability to build a batch from files "matched files" on the printers. That would require some method of vailidating file versions which feels "dirty".

thatguy-jaysenodell, 2025-03-25:
### Some background
Some prereq understanding for the screen shots that are included... I'm mainsail but ... I like to do things "my way". Here's my system arch so we can undertsand it a bit:
* Printers are ender5+s (A and B) fronted pi3bs by running deb bookworm with Klipper and Moonraker. Nothing else. 
* Docker server (cncstation.jodell.net:80) hosts Mainsail operating in remote mode. 
* Docker server (cncstation.jodell.net:4000) hosts FDMMonser using latest container image.
* I've created some simple test files with a G28 for use in this fun and dumped my current file tree from moonraker. If you gcode files higher than 4, I just ran that for loop with bigger numbers. 

[image]

[image]

### Longwinded answers in an effort to provide all the information you need
In order of asking:
* Yes. I will refer to it as "batch update" going forward.
* Upload, to each printer, yes. Visable under each printer file list in FDMM, yes (verified but not shown).  Not visible for batch reprint 

[image]

[image]

[image]
 
* See the screen shot immediately above... there is no record of the batch and no execution of the file. From an FDMM perspetive I can not trigger "a batch" for execution of the just uploaded file simultaneously on muliple printers. IN addition, the printers only show "upload" and no print execution. From the moonraker.log on e5b

`2025-03-25 09:38:14,008 [application.py:prepare()] - Upload Request Received from 192.168.0.99
Content-Type: multipart/form-data; boundary=--------------------------237088041080347859673675
2025-03-25 09:38:14,024 [application.py:post()] - Processing Uploaded File: 1.gcode
2025-03-25 09:38:14,544 [shell_command.py:pipe_data_received()] - INFO:metadata:Processor preprocess_cancellation is disabled

2025-03-25 09:38:14,651 [shell_command.py:_check_proc_success()] - Command (/home/klipper/moonraker-env/bin/python /home/klipper/moonraker/moonraker/components/file_manager/metadata.py -c /tmp/metacfg-voq6ebb9.json) successfully finished
2025-03-25 09:38:14,657 [application.py:log_request()] - 201 POST /server/files/upload (192.168.0.99) [_TRUSTED_USER_] 650.26ms
2025-03-25 09:38:14,676 [file_manager.py:get_file_list()] - Updating File List <gcodes>...
2025-03-25 09:39:17,854 [file_manager.py:get_file_list()] - Updating File List <gcodes>...
2025-03-25 09:46:26,182 [file_manager.py:get_file_list()] - Updating File List <gcodes>...`

* The start can be from either FDMM or the printer. The below screeshots I used FDMM for e5a, moonraker for e5b then did the batch reprint. 

[image]

[image]

[image]

[image]

Also note that the previous batch jobs are deleted. Here you can see i did the same process with 2.gcode and post manual kickoff of the prints, the 1.gcode batch is gone. 

[image]

When I submit the print request for the 2.gcode, it fires off properly. 

### The problem description:
Add... 

nor is a FDMM batch job history created to manually start the batch print.

davidzwa, 2025-03-28:
@all-contributors add @thatguy-jaysenodell for bug

closed 2025-05-11 by davidzwa