fdm-monster/fdm-monster #4345: [Feature]: Add FDM Monster to Proxmox via Community Scripts.

tadatin, 2025-05-20:
### Checklist

- [x] I agree to follow our [Code of Conduct](https://github.com/fdm-monster/fdm-monster/blob/develop/CODE_OF_CONDUCT.md).
- [x] I have verified no [other issues](https://github.com/fdm-monster/fdm-monster/issues) exist related to my request.

### Is Your Feature Request Related To A Problem? Please describe.

Docker is a very good installation option, but some users prefer Proxmox virtualisation platform, where native container is LXC. Installing Docker on it is possible, but quite complicate to many users. man

### Describe The Solution. Why is it needed?

Collaborate with [Community-scripts](https://community-scripts.github.io/ProxmoxVE/) and create a script to install FDM Monster with simple script, hosted on Proxmox Community Scripts github.
This will provide FDM Monster to additional users who don't want to install on Pi, but already have a Proxmox server and want to use ir for managing printfarm.

### Additional Context

Octoprint is there, so IMHO, it can be doable with FDM Monster: https://community-scripts.github.io/ProxmoxVE/scripts?id=octoprint

davidzwa, 2025-05-21:
Thanks for bringing this up! I think its a great addition to the installation options.

I find the OctoPrint example easy to understand and I think we can definitely plan and execute this work soon.

Ill follow up once we can say more.

tadatin, 2025-10-05:
Any updates on the situation?